r9y9/pylibfreenect2 #9: Cannot succeed to install

sfornengo, 2016-03-29:
I have installed libfreenect2 in my home directory.
This create freenect2 in my home directory too, containing /include and /lib
Protonect, NiViewer2 and Nite2 are working.
Try installing pylibfreenect2 with:
sudo LIBFREENECT2_INSTALL_PREFIX=/home/stf/freenect2/ pip install git+https://github.com/r9y9/pylibfreenect2
getting errors:

Downloading/unpacking git+https://github.com/r9y9/pylibfreenect2
  Cloning https://github.com/r9y9/pylibfreenect2 to /tmp/pip-3jVmAQ-build
  Running setup.py (path:/tmp/pip-3jVmAQ-build/setup.py) egg_info for package from git+https://github.com/r9y9/pylibfreenect2
    Compiling pylibfreenect2/libfreenect2.pyx because it changed.
    Cythonizing pylibfreenect2/libfreenect2.pyx
    
Requirement already satisfied (use --upgrade to upgrade): numpy>=1.7.0 in /usr/local/lib/python2.7/dist-packages (from pylibfreenect2==0.0.1-dev)
Requirement already satisfied (use --upgrade to upgrade): enum34 in /usr/local/lib/python2.7/dist-packages (from pylibfreenect2==0.0.1-dev)
Installing collected packages: pylibfreenect2
  Running setup.py install for pylibfreenect2
    building 'pylibfreenect2.libfreenect2' extension
    x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/local/lib/python2.7/dist-packages/numpy/core/include -I/home/stf/freenect2/include/libfreenect2 -I/usr/include/python2.7 -c pylibfreenect2/libfreenect2.cpp -o build/temp.linux-x86_64-2.7/pylibfreenect2/libfreenect2.o -std=c++11
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
    In file included from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1781:0,
                     from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                     from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
                     from pylibfreenect2/libfreenect2.cpp:287:
    /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
     #warning "Using deprecated NumPy API, disable it by " \
      ^
    In file included from pylibfreenect2/libfreenect2.cpp:298:0:
    /home/stf/freenect2/include/libfreenect2/frame_listener_impl.h:35:43: fatal error: libfreenect2/frame_listener.hpp: Aucun fichier ou dossier de ce type
     #include <libfreenect2/frame_listener.hpp>
                                               ^
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    Complete output from command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-3jVmAQ-build/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-Om021S-record/install-record.txt --single-version-externally-managed --compile:
    running install

running build

running build_py

creating build

creating build/lib.linux-x86_64-2.7

creating build/lib.linux-x86_64-2.7/pylibfreenect2

copying pylibfreenect2/__init__.py -> build/lib.linux-x86_64-2.7/pylibfreenect2

running build_ext

building 'pylibfreenect2.libfreenect2' extension

creating build/temp.linux-x86_64-2.7

creating build/temp.linux-x86_64-2.7/pylibfreenect2

x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/local/lib/python2.7/dist-packages/numpy/core/include -I/home/stf/freenect2/include/libfreenect2 -I/usr/include/python2.7 -c pylibfreenect2/libfreenect2.cpp -o build/temp.linux-x86_64-2.7/pylibfreenect2/libfreenect2.o -std=c++11

cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]

In file included from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1781:0,

                 from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,

                 from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/arrayobject.h:4,

                 from pylibfreenect2/libfreenect2.cpp:287:

/usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]

 #warning "Using deprecated NumPy API, disable it by " \

  ^

In file included from pylibfreenect2/libfreenect2.cpp:298:0:

/home/stf/freenect2/include/libfreenect2/frame_listener_impl.h:35:43: fatal error: libfreenect2/frame_listener.hpp: Aucun fichier ou dossier de ce type

 #include <libfreenect2/frame_listener.hpp>

                                           ^

compilation terminated.

error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-3jVmAQ-build/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-Om021S-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-3jVmAQ-build
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.4', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 235, in main
    return command.main(cmd_args)
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 161, in main
    text = '\n'.join(complete_log)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 42: ordinal not in range(128)

r9y9, 2016-03-30:
> /home/stf/freenect2/include/libfreenect2/frame_listener_impl.h:35:43: fatal error: libfreenect2/frame_listener.hpp: Aucun fichier ou dossier de ce type

I can reproduce the problem; libfreenect2 headers wasn't in header search paths with non-default libfreenect2 installation path. #10 should fix this. I will merge #10 when it's green.

For the record, the exact error log I saw was:

```
% LIBFREENECT2_INSTALL_PREFIX=~/freenect2 python setup.py develop   
/Users/ryuyamamoto/anaconda/envs/py35/lib/python3.5/site-packages/setuptools-20.1.1-py3.5.egg/setuptools/dist.py:285: UserWarning: Normalizing '0.0.1-dev' to '0.0.1.dev0'
running develop
running egg_info
writing requirements to pylibfreenect2.egg-info/requires.txt
writing dependency_links to pylibfreenect2.egg-info/dependency_links.txt
writing top-level names to pylibfreenect2.egg-info/top_level.txt
writing pylibfreenect2.egg-info/PKG-INFO
reading manifest file 'pylibfreenect2.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'pylibfreenect2.egg-info/SOURCES.txt'
running build_ext
building 'pylibfreenect2.libfreenect2' extension
gcc -fno-strict-aliasing -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Users/ryuyamamoto/anaconda/envs/py35/include -arch x86_64 -I/Users/ryuyamamoto/anaconda/envs/py35/lib/python3.5/site-packages/numpy/core/include -I/Users/ryuyamamoto/freenect2/include -I/Users/ryuyamamoto/a
naconda/envs/py35/include/python3.5m -c pylibfreenect2/libfreenect2.cpp -o build/temp.macosx-10.5-x86_64-3.5/pylibfreenect2/libfreenect2.o -std=c++11 -stdlib=libc++ -mmacosx-version-min=10.8
In file included from pylibfreenect2/libfreenect2.cpp:283:
In file included from /Users/ryuyamamoto/anaconda/envs/py35/lib/python3.5/site-packages/numpy/core/include/numpy/arrayobject.h:4:
In file included from /Users/ryuyamamoto/anaconda/envs/py35/lib/python3.5/site-packages/numpy/core/include/numpy/ndarrayobject.h:18:
In file included from /Users/ryuyamamoto/anaconda/envs/py35/lib/python3.5/site-packages/numpy/core/include/numpy/ndarraytypes.h:1781:
/Users/ryuyamamoto/anaconda/envs/py35/lib/python3.5/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-W#warnings]
#warning "Using deprecated NumPy API, disable it by " \
 ^
pylibfreenect2/libfreenect2.cpp:293:10: fatal error: 'frame_listener.hpp' file not found
#include "frame_listener.hpp"
         ^
1 warning and 1 error generated.
```

r9y9, 2016-03-30:
Feel free to reopen if you still see the problem.

closed 2016-03-30 by r9y9

sfornengo, 2016-03-30:
Ok, this fix installation problem.
But now running tests failed:
nosetests -v -w tests/

Failure: ImportError (libfreenect2.so.0.2: cannot open shared object file: No such file or directory) ... ERROR

======================================================================
ERROR: Failure: ImportError (libfreenect2.so.0.2: cannot open shared object file: No such file or directory)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/nose/loader.py", line 418, in loadTestsFromName
    addr.filename, addr.module)
  File "/usr/local/lib/python2.7/dist-packages/nose/importer.py", line 47, in importFromPath
    return self.importFromDir(dir_path, fqname)
  File "/usr/local/lib/python2.7/dist-packages/nose/importer.py", line 94, in importFromDir
    mod = load_module(part_fqname, fh, filename, desc)
  File "/home/stf/pylibfreenect2-master/tests/test_libfreenect2.py", line 8, in <module>
    from pylibfreenect2 import Freenect2, SyncMultiFrameListener
  File "/usr/local/lib/python2.7/dist-packages/pylibfreenect2/__init__.py", line 68, in <module>
    from .libfreenect2 import *
ImportError: libfreenect2.so.0.2: cannot open shared object file: No such file or directory

----------------------------------------------------------------------
Ran 1 test in 0.038s

FAILED (errors=1)

r9y9, 2016-03-30:
I think you need to set `LD_LIBRARY_PATH` so that `libfreenect2.so.0.2` can be found at runtime. See https://github.com/r9y9/pylibfreenect2/blob/master/.travis.yml#L31 for example. 

Maybe `export LD_LIBRARY_PATH=$HOME/freenect2/lib:$LD_LIBRARY_PATH`?

sfornengo, 2016-03-30:
You are right, thanks !

r9y9, 2016-03-30:
No worries, thanks for your patience!

kaygudo, 2016-08-04:
Hey!
I am too getting the same error when i try to execute "python test.py".Following is the error:
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    import pyfreenect2
  File "/home/kaygudo/apps/pyfreenect2/pyfreenect2.py", line 1, in <module>
    import _pyfreenect2
  File "build/bdist.linux-x86_64/egg/_pyfreenect2.py", line 7, in <module>
  File "build/bdist.linux-x86_64/egg/_pyfreenect2.py", line 6, in __bootstrap__
ImportError: libfreenect2.so.0.2: cannot open shared object file: No such file or directory

  and i tried your method "export LD_LIBRARY_PATH=$HOME/freenect2/lib:$LD_LIBRARY_PATH" but still getting the same error.any suggestions?
Thanks

r9y9, 2016-08-04:
Where did you install libfreenect2? Does ~/freenect2/lib/libfreenect2.so exist?

kaygudo, 2016-08-04:
Yes, it does

kaygudo, 2016-08-04:
do you know what is the problem?

r9y9, 2016-08-04:
Not sure exactly, but it looks like python failed to load libfreenect2.so dependencies. Please make sure that all dependent libraries are in library search paths. 

```
$ ldd  ~/freenect2/lib/libfreenect2.so
```

may help to find which library are not property loaded.

kaygudo, 2016-08-04:
When i execute the command "ldd  ~/freenect2/lib/libfreenect2.so" my output is:

	linux-vdso.so.1 =>  (0x00007fff0735b000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f007bef8000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f007bcda000)
	libva-drm.so.1 => /usr/lib/x86_64-linux-gnu/libva-drm.so.1 (0x00007f007bad6000)
	libva.so.1 => /usr/lib/x86_64-linux-gnu/libva.so.1 (0x00007f007b8ba000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f007b665000)
	libturbojpeg.so.0 => /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 (0x00007f007b40a000)
	libglfw.so.3 => /usr/lib/x86_64-linux-gnu/libglfw.so.3 (0x00007f007b1f4000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1 (0x00007f007af65000)
	libOpenCL.so.1 => /usr/lib/x86_64-linux-gnu/libOpenCL.so.1 (0x00007f007ad59000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f007aa55000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f007a74f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f007a389000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f007a178000)
	/lib64/ld-linux-x86-64.so.2 (0x000055eecb5e5000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007f0079f6a000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0079d65000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f0079b5d000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f0079828000)
	libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f007961d000)
	libXinerama.so.1 => /usr/lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007f007941a000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f007920a000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f0079003000)
	libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007f0078df9000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f0078bcf000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007f00789a0000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f007878e000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f007858b000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f0078384000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f0078182000)
	libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007f0077f6b000)
	libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007f0077d65000)
	libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007f0077b62000)
	libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007f007795f000)
	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007f0077758000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f0077539000)
	libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007f0077337000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f0077120000)
	libcgmanager.so.0 => /lib/x86_64-linux-gnu/libcgmanager.so.0 (0x00007f0076f05000)
	libnih.so.1 => /lib/x86_64-linux-gnu/libnih.so.1 (0x00007f0076cec000)
	libnih-dbus.so.1 => /lib/x86_64-linux-gnu/libnih-dbus.so.1 (0x00007f0076ae2000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f007689d000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f0076692000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f007648e000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f0076287000)

i am new to linux. can you please tell me what is wrong?

r9y9, 2016-08-04:
might help:

```
export LD_LIBRARY_PATH=$HOME/freenect2/lib:/usr/lib:/lib
```

If it fail, I have no idea. Do you have chance to try with default path libfreenect2 installation instead of custom path installation?

kaygudo, 2016-08-04:
Actually when i follwed the mentioned method to install, which is:
Git clone https://github.com/remexre/pyfreenect2
& sudo python setup.py install
i got following error:
In file included from pyfreenect2.cpp:1:0:
pyfreenect2.hpp:5:41: fatal error: libfreenect2/libfreenect2.hpp: No such file or directory
#include
^
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

thus i followed this link : http://kubotti.hatenablog.com/entry/2016/05/30/230910
in which, in the file setup.py i changed : 
include_dirs=[numpy.get_include(),"/usr/local/freenect2/include"]
from:
include_dirs=[numpy.get_include()]
which solved my error.
got any clue now?
Thanks

r9y9, 2016-08-04:
Ah, pyfreenect2 is NOT pylibfreenect2. Please file an issue to https://github.com/remexre/pyfreenect2 for the problem. If you want to try pylibfreenect2 instead, please read http://r9y9.github.io/pylibfreenect2/installation.html.

kaygudo, 2016-08-04:
Hey! i realized that and tried to git cloned your repo and executed the command "python setup.py install". But i get this error:
Traceback (most recent call last):
  File "setup.py", line 25, in <module>
    raise OSError("{}: is not found".format(libfreenect2_configh_path))
OSError: /usr/local/include/libfreenect2/config.h: is not found

r9y9, 2016-08-04:
You need to set `LIBFREENECT2_INSTALL_PREFIX` as the installation [doc](http://r9y9.github.io/pylibfreenect2/installation.html#building-pylibfreenect2) says. Before running setup.py, please make sure that:

> You have installed libfreenect2 correctly and confirm that Protonect works.
> You have set LIBFREENECT2_INSTALL_PREFIX environmental variable (default: /usr/local/).

kaygudo, 2016-08-05:
Yes, my protonect and everything is working fine.But i am still getting this error while installation.Also one thing which i have noticed is that, in your repo you have an file named ".travis.yml" but when i do git clone it changes to "appveyor.yml".

r9y9, 2016-08-05:
The repo contains both .travis.yml and appveyor.yml. `LIBFREENECT2_INSTALL_PREFIX`?

kaygudo, 2016-08-05:
Hey! i am new to linux please can you tell me how to set LIBFREENECT2_INSTALL_PREFIX? i think i have set it already but still to confirm if you can.Also  when i git clone only appveyor.yml i can see no .travis.yml

r9y9, 2016-08-05:
Ok, assuming you have installed libfreenect2 into ~/freenect2, try the following one line:

```
LIBFREENECT2_INSTALL_PREFIX=~/freenect2 python setup.py install
```

> Also when i git clone only appveyor.yml i can see no .travis.yml

Files start with dot (.) prefix such as .travis.yml are secret files. You can find that files with `ls -a` command, which `-a` means "show secret files".

kaygudo, 2016-08-05:
Hey! thanks that did the installation part, actually i have freenect2 in home directory as well as in /usr/local. But then when i tried to run "nosetests -v -w tests/"  i get following error:

ImportError: No module named libfreenect2

----------------------------------------------------------------------
Ran 1 test in 0.001s

FAILED (errors=1)

also if i tried to execute "multiframe_listener.py" by typing "python multiframe_listener.py" i get the same old error:

ImportError: libfreenect2.so.0.2: cannot open shared object file: No such file or directory

r9y9, 2016-08-05:
Check if ~/freenect2/lib is in LD_LIBRARY_PATH:

```
echo $LD_LIBRARY_PATH
```

if not, try:

```
export LD_LIBRARY_PATH=$HOME/freenect2/lib:$LD_LIBRARY_PATH
```

and then:

```
python multiframe_listener.py
```

kaygudo, 2016-08-05:
Hey! i tried but nopes, still the same error:

Traceback (most recent call last):
  File "multiframe_listener.py", line 6, in <module>
    from pylibfreenect2 import Freenect2, SyncMultiFrameListener
  File "build/bdist.linux-x86_64/egg/pylibfreenect2/__init__.py", line 108, in <module>
  File "build/bdist.linux-x86_64/egg/pylibfreenect2/libfreenect2.py", line 7, in <module>
  File "build/bdist.linux-x86_64/egg/pylibfreenect2/libfreenect2.py", line 6, in __bootstrap__
ImportError: libfreenect2.so.0.2: cannot open shared object file: No such file or directory

kaygudo, 2016-08-05:
First i tried but it said that : failed to open kinect.Then i upgraded my usb from usb 2.0 to usb 3.0 which solved the "failed to open kinect" , but then again i got this error.

m6c7l, 2016-09-15:
hey! putting LD_LIBRARY_PATH into ~/.profile, ~/.bashrc etc. is not working.
best practise for me in such a case is just to link or copy the required library into python libs directory:
e.g. > ln -s ~/openkinect/freenect2/lib/libfreenect2.so.0.2 ~/python/3.5.2/lib
this should work for you as well. regards

evancsj, 2017-03-29:
when i run
`python multiframe_listener.py`

i got
`Traceback (most recent call last):
  File "multiframe_listener.py", line 6, in <module>
    from pylibfreenect2 import Freenect2, SyncMultiFrameListener
  File "build/bdist.linux-x86_64/egg/pylibfreenect2/__init__.py", line 108, in <module>
  File "build/bdist.linux-x86_64/egg/pylibfreenect2/libfreenect2.py", line 7, in <module>
  File "build/bdist.linux-x86_64/egg/pylibfreenect2/libfreenect2.py", line 6, in __bootstrap__
ImportError: /home/csj/.cache/Python-Eggs/pylibfreenect2-0.1.2.dev0-py2.7-linux-x86_64.egg-tmp/pylibfreenect2/libfreenect2.so: undefined symbol: _ZN12libfreenect29Freenect221getDeviceSerialNumberB5cxx11Ei
`

can anyone tell me what's wrong with it?

thanks

r9y9, 2017-03-30:
@evancsj I wonder the problem is not pylibfreenect2 specific. Stackoverflow might be a good place to ask. If you need a help from me, please open a new issue with detailed steps to reproduce.

adilbonzi, 2017-04-02:
@r9y9 for setting LIBFREENECT2_INSTALL_PREFIX, the freenect2 folder doesn not consists of any setup.py file. In my freenect2, in include's folder i have the config.h file, but when i run this command;
LIBFREENECT2_INSTALL_PREFIX=~/freenect2 python setup.py install, i receive the error no such file or directory. Any suggestions?

r9y9, 2017-04-02:
@adilbonzi Please paste the full log message. It should say the exact location that the setup.py is looking for.  https://github.com/r9y9/pylibfreenect2/blob/e43c0b8106e10e0b7dad9f94c8d8ce5534895295/setup.py#L25.

adilbonzi, 2017-04-03:
@r9y9 this is the error message i receive. My libfreenect2 is in my home folder. 

adil@adil-Inspiron-5520:~$ LIBFREENECT2_INSTALL_PREFIX=~/freenect2 python setup.py install
python: can't open file 'setup.py': [Errno 2] No such file or directory

r9y9, 2017-04-03:
@adilbonzi You need to clone the pylibfreenect2 repository:

```
git clone https://github.com/r9y9/pylibfreenect2
cd pylibfreenect2
LIBFREENECT2_INSTALL_PREFIX =/freenect2 python setup.py install
```

adilbonzi, 2017-04-03:
adil@adil-Inspiron-5520:~/pylibfreenect2$ LIBFREENECT2_INSTALL_PREFIX=~/freenect2 python setup.py install
Traceback (most recent call last):
  File "setup.py", line 61, in <module>
    raise RuntimeError("Cython is required to generate C++ codes.")
RuntimeError: Cython is required to generate C++ codes.

r9y9, 2017-04-03:
@adilbonzi http://r9y9.github.io/pylibfreenect2/latest/installation.html#build-requirements 

I recommend you to use https://www.continuum.io/downloads to install requirements.

adilbonzi, 2017-04-03:
@r9y9 i did sudo pip install cython and it successfully installed. Then i ran the following commands; 

adil@adil-Inspiron-5520:~/pylibfreenect2$ LIBFREENECT2_INSTALL_PREFIX=~/freenect2 python setup.py install
Compiling pylibfreenect2/libfreenect2.pyx because it changed.
[1/1] Cythonizing pylibfreenect2/libfreenect2.pyx
running install
error: can't create or remove files in install directory

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/test-easy-install-3076.write-test'

The installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /usr/local/lib/python2.7/dist-packages/

Perhaps your account does not have write access to this directory?  If the
installation directory is a system-owned directory, you may need to sign in
as the administrator or "root" account.  If you do not have administrative
access to this machine, you may wish to choose a different installation
directory, preferably one that is listed in your PYTHONPATH environment
variable.

For information on other options, you may wish to consult the
documentation at:

  https://pythonhosted.org/setuptools/easy_install.html

Please make the appropriate changes for your system and try again.

PR #10 (2016-03-30): Fix header search path to the top of libfreenect2 include directory
this should fix #9
commit 373eb3e7 (2016-03-30): Fix header search path to the top of libfreenect2 include directory

this should fix #9