r9y9/pylibfreenect2 #12: missing cimport in module 'utility': /usr/lib/python2.7/dist-packages/Cython/Includes/libcpp/map.pxd

r9y9, 2016-04-01:
Perhaps depends on cython version. Previously @karanbudhraja reported this issue. From his email, 

```
sudo python setup.py install
missing cimport in module 'utility': /usr/lib/python2.7/dist-packages/Cython/Includes/libcpp/map.pxd
running install
running bdist_egg
running egg_info
writing requirements to pylibfreenect2.egg-info/requires.txt
writing pylibfreenect2.egg-info/PKG-INFO
writing top-level names to pylibfreenect2.egg-info/top_level.txt
writing dependency_links to pylibfreenect2.egg-info/dependency_links.txt
reading manifest file 'pylibfreenect2.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'pylibfreenect2.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/pylibfreenect2
copying build/lib.linux-x86_64-2.7/pylibfreenect2/__init__.py -> build/bdist.linux-x86_64/egg/pylibfreenect2
copying build/lib.linux-x86_64-2.7/pylibfreenect2/libfreenect2.so -> build/bdist.linux-x86_64/egg/pylibfreenect2
byte-compiling build/bdist.linux-x86_64/egg/pylibfreenect2/__init__.py to __init__.pyc
creating stub loader for pylibfreenect2/libfreenect2.so
byte-compiling build/bdist.linux-x86_64/egg/pylibfreenect2/libfreenect2.py to libfreenect2.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying pylibfreenect2.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pylibfreenect2.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pylibfreenect2.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pylibfreenect2.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pylibfreenect2.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating 'dist/pylibfreenect2-0.0.1_dev-py2.7-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing pylibfreenect2-0.0.1_dev-py2.7-linux-x86_64.egg
Copying pylibfreenect2-0.0.1_dev-py2.7-linux-x86_64.egg to /usr/local/lib/python2.7/dist-packages
Adding pylibfreenect2 0.0.1-dev to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/pylibfreenect2-0.0.1_dev-py2.7-linux-x86_64.egg
Processing dependencies for pylibfreenect2==0.0.1-dev
Searching for enum34==1.1.2
Best match: enum34 1.1.2
Processing enum34-1.1.2-py2.7.egg
enum34 1.1.2 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/enum34-1.1.2-py2.7.egg
Searching for numpy==1.8.2
Best match: numpy 1.8.2
numpy 1.8.2 is already the active version in easy-install.pth

Using /usr/lib/python2.7/dist-packages
Finished processing dependencies for pylibfreenect2==0.0.1-dev
```

I think it's worth trying the following snippet:

```cython
import numpy as np
cimport numpy as np
cimport cython

from libcpp.map cimport map

def f(map[int,int] m):
    for pair in m:
        print(pair.first, pair.second)
```

```
$ cython -v --cplus test.pyx
```

to see if the issue is caused by cython or pylibfreenect2.

bafu, 2016-09-10:
@r9y9 I also met this issue with Cython 0.20.1 in Ubuntu 14.04.  After installed Cython 0.24.1 via Pip, this issue was resolved.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.1 LTS
Release:	14.04
Codename:	trusty

$ apt-cache policy cython
cython:
  Installed: 0.20.1+git90-g0e6e38e-1ubuntu2
  Candidate: 0.20.1+git90-g0e6e38e-1ubuntu2
  Version table:
 *** 0.20.1+git90-g0e6e38e-1ubuntu2 0
        500 http://ports.ubuntu.com/ubuntu-ports/ trusty/main armhf Packages
        100 /var/lib/dpkg/status

$ pip list | grep Cython
Cython (0.24.1)
```

r9y9, 2016-09-10:
@bafu Thanks for the report! Ok, now I think I find the problem; the error was caused because absolute `cimport` in `Cython/Includes/libcpp/map.pxd` was not working property.  See fix for https://github.com/cython/cython/commit/59da9514bbb11d20e23e07c1d632a68d20186a36. Will fix required cython version to v0.21 that includes the fix for this.

commit 5f57a911 (2016-09-10): Requires cython >= 0.21.0

fixes #12
closed 2016-09-10 by r9y9