r9y9/pylibfreenect2 #16: Logging Request

gargoyle3333, 2016-05-05:
Hi, this is a feature request - is there a way to convert print statements to use loggers instead? This would allow me to disable output completely or redirect output to a file more easily.

r9y9, 2016-05-05:
There is no way to do that for now, but it should be definitely be possible. Will try to implement if I get a change.

r9y9, 2016-05-07:
@gargoyle3333 I implemented logging utilities of libfreenect2. You can change logger level by creating a new logger:

```py
from pylibfreenect2 import createConsoleLogger, setGlobalLogger
from pylibfreenect2 import LoggerLevel

logger = createConsoleLogger(LoggerLevel.Debug)
setGlobalLogger(logger)
```

To disable, 

```py
setGlobalLogger(None)
```

if you want to replace print statements with libfreenecg2's logger, you can try the following:

```py
logger.log(LoggerLevel.Debug, b"message")
```