r9y9/pylibfreenect2 #18: Device gets closed when the code gets out of scope

tianyangma, 2016-07-06:
A sample code snippet is:

```python
def open_device():
    # Create a processing pipeline.
    pipeline = OpenGLPacketPipeline()
    
    # Create a logger.
    logger = createConsoleLogger(LoggerLevel.Debug)
    setGlobalLogger(logger)
                
    # Discover Kinect device.
    fn = Freenect2()
    num_devices = fn.enumerateDevices()
    if num_devices == 0:
        print("No device connected!")
        sys.exit(1)
    
    serial = fn.getDeviceSerialNumber(0)
                
    # Connect to the device.
    device = fn.openDevice(serial, pipeline=pipeline)
                
    # Register listeners.
    listener = SyncMultiFrameListener(FrameType.Depth)
    device.setIrAndDepthFrameListener(listener)
    
    return device, listener

device, listener = open_device()

while True:
    cv2.waitKey(-1)
```

In this case, the device gets closed when exiting the function call ```open_device()```.

r9y9, 2016-07-07:
Yes, you should keep `fn` (instance of `Freenect2`) to keep your device open. This is because `Freenect2` is responsible to manage devices and `openDevice` just returns the raw pointer of allocated device object. In this case, `fn` will get deallocated after `open_device` and device get deallocated (i.e.closed) as well.

tianyangma, 2016-07-07:
Thanks for the help and the great work! Just tried the code below, and it works:

```Python

def open_device():
    # Create a processing pipeline.
    pipeline = OpenGLPacketPipeline()

    # Create a logger.
    logger = createConsoleLogger(LoggerLevel.Debug)
    setGlobalLogger(logger)

    # Discover Kinect device.
    fn = Freenect2()
    num_devices = fn.enumerateDevices()
    if num_devices == 0:
        print("No device connected!")
        sys.exit(1)

    serial = fn.getDeviceSerialNumber(0)

    # Connect to the device.
    device = fn.openDevice(serial, pipeline=pipeline)

    # Register listeners.
    listener = SyncMultiFrameListener(FrameType.Depth)
    device.setIrAndDepthFrameListener(listener)

    return device, listener, fn

device, listener, fn = open_device()

while True:
    cv2.waitKey(-1)

```

closed 2016-07-07 by tianyangma