r9y9/pylibfreenect2 #20: Generation of point clouds

kaygudo, 2016-08-10:
Hello,
Is there a way to generate point clouds using pylibfreenect2?There is a repo based on libfreenect2 by the name "libfreenect2pclgrabber" can we use this?

r9y9, 2016-08-17:
It would need https://github.com/r9y9/pylibfreenect2/pull/19 at least, though I don't know a good way to visualize point clouds in python.  Actually a half year ago I tried to wrap PCL (that includes PCLVisualizer) in cython (https://github.com/r9y9/pypcl) but end up with stopping development, because of difficulty of exposing C++ templated classes to pytnon.

r9y9, 2016-10-02:
#23

kaygudo, 2016-10-07:
I managed to generate point clouds using depth and registered frames! :) thanks anyways

r9y9, 2016-10-09:
#19 should provides functionalities to generate point clouds. Will close this. Feel free to reopen if you have any issue related to this.

closed 2016-10-09 by r9y9