r9y9/pylibfreenect2 #21: Packet Pipelines

rascustoms, 2016-10-01:
I am unable to get either OpenCL or OpenGL packet pipelines to import in the example script. Only the CPU pipeline will import. Both pipelines will run fine in libfreenect2 Protonect test program. 

System setup is:
- Windows 10
- Python 3.5.2 64bit
- Cython 0.24
- libfreenect and pylibfreenect both built from master branches (Also tried pylibfreenect from pipi)

Thank you

r9y9, 2016-10-02:
OpenCL and OpenGL packet pipelines in pylibfrenect2 should be automatically enabled if you have built libfreenect2 with those enabled. So it seems that https://github.com/r9y9/pylibfreenect2/blob/5f57a91144e39f01ff119079d82ecafeb4ef8df9/setup.py#L64-L78 isn't working on windows, probably due to use of `cat` and `grep` (which I guess doesn't exist on your environment). Since I don't have windows environment right now, it would be great if you could find a windows workaround for `has_define_in_config` working property.

rascustoms, 2016-10-02:
I've added a windows workaround. Also I noticed this library is very hard to find (through google), but yet seems to be one of the only active ways of using Kinect for Windows 2 with Open CV. Hopefully it can gain some more notice. Thank you for your help.

commit 5e8bd338 (2016-10-02): Merge pull request #22 from rascustoms/master

Fixes #21
closed 2016-10-02 by r9y9