r9y9/pylibfreenect2 #28: Installation errors

pranali3215, 2016-11-15:
While executing the pip command I get the following error:

OSError: /home/abc/libfreenect2/include/libfreenect2/config.h: is not found

This is because the file is named config.h.in and not config.h
Even if I change the filename, during installation it is not able to find the lib folder.
This is because the lib folder is inside the build folder.

Is there something I am missing for doing this installation ?

r9y9, 2016-11-15:
Did you follow the installation guide? http://r9y9.github.io/pylibfreenect2/installation.html#building-pylibfreenect2

pranali3215, 2016-11-15:
Yes! 
My Protonect works and my LIBFREENECT2_INSTALL_PREFIX points to the correct location.

r9y9, 2016-11-15:
umm

> This is because the file is named config.h.in and not config.h

I think `config.h.in` is compiled into `config.h` during compilation of libfreenect2. So If you successfully built libfreenect2, you will have the `config.h` in your build tree.

pranali3215, 2016-11-15:
After compilation, a new folder freenect2 got created which has the config.h file. 
That folder has the include and lib folders. However, the libfreenect2 folder is as it is.

r9y9, 2016-11-15:
You have to point out the directory `freenect2`, not `libfreenect2`.

Maybe

```
export LIBFREENECT2_INSTALL_PREFIX=~/freenect2
```

r9y9, 2016-11-15:
https://github.com/r9y9/pylibfreenect2/issues/9

pranali3215, 2016-11-15:
OSError: /usr/local/include/libfreenect2/config.h: is not found

Same error :(

r9y9, 2016-11-15:
> After compilation, a new folder freenect2 got created which has the config.h file. 

Please point to the directory that config.h exists before installation.

pranali3215, 2016-11-15:
There is no config.h before installation. As I said, its config.h.in. 
Only after installation does it become config.h

r9y9, 2016-11-15:
http://r9y9.github.io/pylibfreenect2/installation.html#building-pylibfreenect2

> Please make sure that:
> 
> You have installed libfreenect2 correctly and confirm that Protonect works.

Following the step, you will have libfreenect2 installed, i.e. you should have config.h before `pylibfreenect2` installation. pylibfreenect2 is not libfreenect2.

pranali3215, 2016-11-15:
As I said before, the config.h file is in a folder named freenect2 which is in /usr/local.
When I try to install the pylibfreenect2, it is not able to find the config.h file even though the correct location is given.

I am not confusing pylibfreenect2 with libfreenect2 as I do not have the pylibfreeect2 folder on my system. Hence, no confusion.

r9y9, 2016-11-15:
Could you please tell me the full path of the config.h you have? You mean `/usr/local/include/freenect2/config.h`?

pranali3215, 2016-11-15:
/usr/local/freenect2/include/libfreenect2/config.h

r9y9, 2016-11-15:
Can you try the following?

```
export LIBFREENECT2_INSTALL_PREFIX=/usr/local/freenect2
```

pranali3215, 2016-11-15:
OSError: /usr/local/include/libfreenect2/config.h: is not found

:(

pranali3215, 2016-11-16:
> $ echo $LD_LIBRARY_PATH

**/usr/local/freenect2/lib**

  > $ echo $LIBFREENECT2_INSTALL_PREFIX

**/usr/local/freenect2**

Output: OSError: /usr/local/include/libfreenect2/config.h: is not found

r9y9, 2016-11-16:
```
LIBFREENECT2_INSTALL_PREFIX=/usr/local/freenect2 pip install pylibfreenect2
```

Doesn't it also work?

pranali3215, 2016-11-16:
Hey,
So I cloned the repository and made following changes in the setup.py file:

libfreenect2_install_prefix = os.environ.get( "LIBFREENECT2_INSTALL_PREFIX", "/usr/local/")
to
libfreenect2_install_prefix = os.environ.get("LIBFREENECT2_INSTALL_PREFIX", "/usr/local/freenect2/")

After these changes, the setup.py works and the multiframe_listener.py is also working.

What I have understood is that it is not recognising LIBFREENECT2_INSTALL_PREFIX and hence it is taking the default, which is /usr/local/.
Thus the error that was coming was /usr/local/include as the default path did not have /usr/local/freenect2 in its path.

r9y9, 2016-11-16:
Yes, that change should work. `LIBFREENECT2_INSTALL_PREFIX=/usr/local/freenect2 pip install pylibfreenect2` should also work as well though. I guess environmental variable `LIBFREENECT2_INSTALL_PREFIX` isn't propagated during installation time.

pranali3215, 2016-11-16:
I guess so. 
Thank you for your help and patience !!!
I appreciated your prompt answers.

closed 2016-11-16 by pranali3215