r9y9/pylibfreenect2 #29: Performance Difference

evancloutier, 2016-11-23:
Hi!

I'm noticing a performance difference between the ./bin/Protonect example with libfreenect2 and the multiframe listener example you've provided. Is this expected? The refresh rate is not very smooth. I'm running everything on macOS El Capitan.

r9y9, 2016-11-23:
What do you mean by performance? 1) GUI refresh rate or 2) packet processing time? 

1) As I noted in [example code](https://github.com/r9y9/pylibfreenect2/blob/5575097e89929e8aa706d8c9e15676dbb2387981/examples/multiframe_listener.py#L67-L70), it's expected if you don't have fast visualization backend.

2) I believe there is not much difference.

r9y9, 2017-04-23:
I think this can be closed. Feel free to reopen if you still have an issue.

closed 2017-04-23 by r9y9