r9y9/pylibfreenect2 #30: ColorFrame to ndarray (Question)

paulomarcos, 2016-12-01:
Is there a way to convert the colorframe to ndarray format?
I am using the following, provided by the tutorial (omitting other types):
```
...
frames = listener.waitForNewFrame()
color = frames["color"]
...
im = color.asarray()
```
I need the color frame to be an ndarray in order to fit in this piece of code:

```
#im (ndarray): a color image in BGR order
im_orig = im.astype(np.float32, copy=True)
im_orig -= cfg.PIXEL_MEANS
```
However, when I try to use _im = color.asarray()_ the program gives the following error:
```
im_orig -= cfg.PIXEL_MEANS
ValueError: operands could not be broadcast together with shapes (540,960) (1,1,3) (540,960) 
```
What currently fits/works in that piece of code is:
```
#im_file is the name of an image in jpg format, i.e. 'demo/cat.jpg'
im = cv2.imread(im_file)
```

Probably this is something very simple, but I am having a hard time trying to solve it.
Is there anyone that could help me?

r9y9, 2016-12-01:
I suspect you are getting dimension mismatch? I'm getting color frame as 4-channel data (whose shape is `(1080, 1920, 4)`) and can subtract mean pixel value:

```python
>>> im = color.asarray()
>>> im.__class__
<class 'numpy.ndarray'>
>>> im.shape
(1080, 1920, 4)
>>> PIXEL_MEANS = np.mean(im, (0,1))
>>> PIXEL_MEANS.shape
(4,)
>>> PIXEL_MEANS
array([ 132.62429784,  137.5282745 ,  135.42699171,  255.        ])
>>> im - PIXEL_MEANS
array([[[  49.37570216,   44.4717255 ,   45.57300829,    0.        ],
        [  49.37570216,   44.4717255 ,   46.57300829,    0.        ],
        [  48.37570216,   44.4717255 ,   45.57300829,    0.        ],
...
```

subtracting alpha-channel is not a good idea though (just a demo).

paulomarcos, 2016-12-02:
@r9y9 Thank you for your response. You made me realize the difference in number of channels, hence I was able to accomplish what I wanted using the OpenCV method cvtColor as in:
`newImage = cv2.cvtColor(im, cv2.COLOR_RGBA2RGB)`
Have a good one.

closed 2016-12-02 by paulomarcos

aegorfk, 2018-08-06:
@r9y9 Hi, Just in case, what do you recommend to do if there is a need to have just 3 channels (RGB)? Is it the only one solution to subtract the alpha-channel? If yes, how to do this in package?