r9y9/pylibfreenect2 #32: Registration has no attribute 'getPointXYZ'

stuartbourne, 2017-01-29:
I've been working with these bindings for a couple months now, and have had few to no problems. However, I am currently trying to perform depth analysis using the getPointXYZ function from the Registration class. This is the error I'm receiving: "AttributeError: 'pylibfreenect2.libfreenect2.Registration' object has no attribute 'getPointXYZ'". Just curious as to if this is a common issue or one native to my machine?

r9y9, 2017-01-29:
If you are using the release version of pylibfreenect2 (v0.1.1), it doesn't support getPointXYZ. Please try the latest master of pylibfreenect2 and libfreenect2.

evancloutier, 2017-01-30:
@r9y9 Is it mandatory to update the libfreenect2 branch to interface with your new master branch? I'm having the same issue of this endpoint being unavailable.

evancloutier, 2017-01-31:
@r9y9 To follow up on this, I've gotten the `getPointXYZ` method to work, but am getting a tuple consisting of `(nan, nan, nan)` when calling from my own class. However, when I use your multiframe_listener.py example, I'm able to get XYZ coordinates.

Do you have any idea as to why this might be? I've followed that example very closely, but cannot find the root of the issue.

Kinect class: https://github.com/evancloutier/490-virtual-piano/blob/key-class/Controller/Kinect/Kinect.py

Main class (where I am calling getPointXYZ): https://github.com/evancloutier/490-virtual-piano/blob/key-class/Controller/Main.py

r9y9, 2017-01-31:
@evancloutier libfreenect2 v0.2.0 is actually enough, but no problem with the master branch since I'm testing against it as well. For the `(nan, nan, nan)`, I assume you are getting zero or nan  in depth? `NaN` can be happen in depth. See https://github.com/OpenKinect/libfreenect2/issues/780 for the related issue.

evancloutier, 2017-01-31:
@r9y9 When I replicate your multiframe_listener.py example, I'm able to get non-zero values for depth while streaming from my Kinect v2. However, when I implement the same methods within my own Kinect class, the values become `(nan, nan, nan)`. When I display the registration, there is nothing visually incorrect with the frames either.

Running your example tells me that there are readable depth values there, but my own implementation isn't picking them up for some reason. Any ideas?

r9y9, 2017-01-31:
https://github.com/evancloutier/490-virtual-piano/blob/af4db63cf6e818d3a299076bf2c1593744ba4bed/Controller/Main.py#L28 Why do you specify `960, 540` for row and col indices? Depth image size is 512x424 so It's out of bound.

r9y9, 2017-01-31:
Forgot to mention, @evancloutier (in case you didn't notice, sorry if not)

evancloutier, 2017-01-31:
@r9y9 That fixed it - wasn't paying attention to the bounds. Thanks!

r9y9, 2017-01-31:
@evancloutier I'm grad to hear that your problem fixed. I will probably add bounds checking into some methods, thanks!

evancloutier, 2017-02-09:
@r9y9 I was looking at the libfreenect2 implementation of the Registration, and was wondering if you wrapped all of the internal methods of the registration?

I'm looking at this particular method for converting a depth point to color point, used by the apply method, which can be found [here](https://github.com/OpenKinect/libfreenect2/blob/master/src/registration.cpp#L87-L106).

r9y9, 2017-02-09:
@evancloutier No, only public APIs are wrapped. Internal classes/methods are hidden in implementation. If you think the particular method is useful, you can request libfreenect2 community to expose it.

r9y9, 2017-04-23:
I think this can be closed. Feel free to reopen if you still have an issue.

closed 2017-04-23 by r9y9