r9y9/pylibfreenect2 #34: Packet buffer is NULL

Dagrol, 2017-04-22:
I am getting the following error on Ubuntu 14.04 while running the examples:

[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
.....
it continuously prints the above and eventually crashes. Am I missing something in the installation? since they worked fine on OSX

r9y9, 2017-04-22:
Do you see the error with `Protonect`?

Dagrol, 2017-04-22:
Nope. Protonect works fine. I get that error as soon as I run the examples and any other scripts I wrote which use pylibfreenect2 which were working fine on OSX. I am starting to wonder if it might be a graphics card issue? (I am running ubuntu 14.04 on a macbook pro with amd Radeon HD 8870M)

r9y9, 2017-04-23:
Um, it might be the issue related graphics, but I'm afraid that I don't have a workaround right now. It's worth trying to see which packet processor triggers this; OpenGLPacketPipeline, OpenCLPacketPipeline or CpuPacketPipeline to dig into the problem.

Dagrol, 2017-04-23:
I am not 100% sure if it's a graphics card issue. It seems very odd that Protonect works but not this. I might be missing some configuration settings

shams3049, 2017-07-24:
How did you resolve this issue? I am also getting recursively same error.

LucasSubli, 2017-09-21:
Any Solutions ? I am having the same issue!

LucasSubli, 2017-09-21:
./Protonect doesn't present any isssues
./Protonect cl DOES present the same error

Any way to disable openCL on pylibfreenect ?

r9y9, 2017-09-22:
@LucasSubli You can switch to CPU or OpenGL packet processor. See:

- https://github.com/r9y9/pylibfreenect2/blob/master/examples/multiframe_listener.py#L11-L21
- http://r9y9.github.io/pylibfreenect2/latest/api.html#packet-pipelines

tokyokuma, 2018-01-31:
I am facing the same problem.
Protonect works normally.
Do you think the problem lies with the GPU?
My PC uses only CPU. Can not I use Kinect V 2 with pylibfreenect with CPU alone?
However,
       from pylibfreenect 2 import CpuPacketPipeline
       pipeline = CpuPacketPipeline ()
I guess that this is programmed to work with only the CPU

I do not know what to do

r9y9, 2018-01-31:
Well, since I don't have Kinect v2 anymore, it's hard to say what is the real cause of this.  Could anybody confirm whether the issue comes from pylibfreenect2 or libfreenect2?  According to the https://github.com/r9y9/pylibfreenect2/issues/34#issuecomment-331284252, there's something wrong with OpenCL. Does Protonect work even with OpenCL backend?

tokyokuma, 2018-02-02:
I tried 
$ ./Protonect gl       It works normally
$ ./Protonect cl　　It does not work properly, Packet buffer is NULL
$ ./Protonect cpu　It works normally
OpenCL doesn't work, but the problem was solved.
I modified the example code(multiframe_listener.py) and tested it.
I made the order of OpenCL the last.
【before】
try:
    from pylibfreenect2 import OpenCLPacketPipeline
    pipeline = OpenCLPacketPipeline()
except:
    try:
        from pylibfreenect2 import OpenGLPacketPipeline
        pipeline = OpenGLPacketPipeline()
    except:
        from pylibfreenect2 import CpuPacketPipeline
pipeline = CpuPacketPipeline()

【after】
try:
    from pylibfreenect2 import OpenGLPacketPipeline
    pipeline = OpenGLPacketPipeline()    
except:
    try:
        from pylibfreenect2 import CpuPacketPipeline
        pipeline = CpuPacketPipeline()
    except:
        from pylibfreenect2 import OpenCLPacketPipeline
        pipeline = OpenCLPacketPipeline()

This code worked without problems!
Thank you for your help!

r9y9, 2018-02-02:
@tokyokuma Thank you very much! So there should be something wrong with OpenCL setup. I used OpenCL backend because it was the fastest backend for me, but I changed example code to use the OpenGL backend instead by default.

leviresende, 2019-05-08:
I had the same issue, so I edited the parameters to:

**depth_method to opengl
reg_method to cpu**

After that, it is working well.

JinXiangLai, 2021-05-02:
> I had the same issue, so I edited the parameters to:
> 
> **depth_method to opengl reg_method to cpu**
> 
> After that, it is working well.

hi, but I still want to use opencl for registration, any idea?

jiapei100, 2024-07-20:
`$ ./bin/Protonect cuda` brought me the same errors:

```zsh
...
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
...
```