r9y9/pylibfreenect2 #36: Installation Issue

yvtheja, 2017-05-19:
Trying to execute the command "pip install pylibfreenect" gives the below error.

```
Downloading/unpacking pylibfreenect2
  Downloading pylibfreenect2-0.1.1.tar.gz (83kB): 83kB downloaded
  Running setup.py (path:/tmp/pip_build_yvtheja/pylibfreenect2/setup.py) egg_info for package pylibfreenect2
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip_build_yvtheja/pylibfreenect2/setup.py", line 112, in <module>
        has_define_in_config("LIBFREENECT2_WITH_OPENCL_SUPPORT"),
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 818, in cythonize
        aliases=aliases)
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 704, in create_extension_list
        for file in nonempty(sorted(extended_iglob(filepattern)), "'%s' doesn't match any files" % filepattern):
      File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 108, in nonempty
        raise ValueError(error_msg)
    ValueError: 'pylibfreenect2/libfreenect2.pyx' doesn't match any files
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip_build_yvtheja/pylibfreenect2/setup.py", line 112, in <module>

    has_define_in_config("LIBFREENECT2_WITH_OPENCL_SUPPORT"),

  File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 818, in cythonize

    aliases=aliases)

  File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 704, in create_extension_list

    for file in nonempty(sorted(extended_iglob(filepattern)), "'%s' doesn't match any files" % filepattern):

  File "/usr/local/lib/python2.7/dist-packages/Cython/Build/Dependencies.py", line 108, in nonempty

    raise ValueError(error_msg)

ValueError: 'pylibfreenect2/libfreenect2.pyx' doesn't match any files

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_yvtheja/pylibfreenect2
Storing debug log for failure in /home/yvtheja/.pip/pip.log

```

I have tried installing OpenCL and installing it again, but it doesn't work. Seems like the error comes from Cython.

Any idea how to fix it?

Thank you :)

r9y9, 2017-05-19:
I can reproduce and found the root cause. It's because libfreenect2.pyx isn't included in the release tarball!

```
pylibfreenect2-0.1.1  tree
.
├── MANIFEST.in
├── PKG-INFO
├── pylibfreenect2
│   ├── __init__.py
│   ├── libfreenect2
│   │   ├── __init__.pxd
│   │   └── libfreenect2.pxd
│   └── libfreenect2.cpp
├── pylibfreenect2.egg-info
│   ├── PKG-INFO
│   ├── SOURCES.txt
│   ├── dependency_links.txt
│   ├── requires.txt
│   └── top_level.txt
├── setup.cfg
├── setup.py
└── tests
    ├── __init__.py
    └── test_libfreenect2.py
```

Sorry about that, you can probably install the package by manually cloning the repo. I'll tag a new release once I have a fix.

r9y9, 2017-05-19:
Ok, updated pypi release, confirmed that it fixes the issue to me.

yvtheja, 2017-05-19:
I have already installed it from the source. Thanks a lot for your help :)

closed 2017-05-19 by yvtheja

zabeehkhan1, 2019-12-13:
Sir i have the same issue do u know how to fix it. i want to install pylibfreenect2 but it give me the following error
Command "python setup.py egg_info" failed with error