r9y9/pylibfreenect2 #38: skipping depth packet. receiving just several frames per second

Negvet, 2017-05-26:
I cloned the repository and installed it using
`python setup.py install`
I am receiving frames with `python multiframe_listener.py`
here is the output of `python multiframe_listener.py`:

> [Info] [Freenect2Impl] enumerating devices...
> [Info] [Freenect2Impl] 8 usb devices connected
> [Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 025159651347
> [Info] [Freenect2Impl] found 1 devices
> [Info] [Freenect2DeviceImpl] opening...
> [Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
> [Info] [Freenect2DeviceImpl] opened
> [Info] [Freenect2DeviceImpl] starting...
> [Debug] [Freenect2DeviceImpl] status 0x090000: 9729
> [Debug] [Freenect2DeviceImpl] status 0x090000: 9731
> [Info] [Freenect2DeviceImpl] submitting rgb transfers...
> [Info] [Freenect2DeviceImpl] submitting depth transfers...
> [Info] [Freenect2DeviceImpl] started
> [Debug] [DepthPacketStreamParser] not all subsequences received 0
> [Debug] [DepthPacketStreamParser] skipping depth packet
> [Debug] [DepthPacketStreamParser] skipping depth packet
> [Debug] [DepthPacketStreamParser] skipping depth packet
> [Debug] [DepthPacketStreamParser] skipping depth packet
> [Debug] [DepthPacketStreamParser] skipping depth packet
> [Debug] [DepthPacketStreamParser] skipping depth packet
> ... so on

But right from the beginning it is missing almost all depth frames, so I could get only several frames per second.
`./bin/Protonect` works just fine.

r9y9, 2017-05-26:
Try commenting out https://github.com/r9y9/pylibfreenect2/blob/288e187ec94fd0df30d483f6c11318820e1d1e06/examples/multiframe_listener.py#L67-L75?

Negvet, 2017-05-26:
I commented out all `cv2.imshow` in `multiframe_listener.py`, but I am still getting a lot of 

> [Debug] [DepthPacketStreamParser] skipping depth packet
> [Debug] [DepthPacketStreamParser] skipping depth packet
> [Debug] [DepthPacketStreamParser] skipping depth packet
> ...

So I did not solve the issue

Negvet, 2017-05-26:
When I run `python selective_streams.py`, I am getting:
> [Info] [DepthPacketStreamParser] 31 packets were lost
> [Info] [DepthPacketStreamParser] 30 packets were lost
> [Info] [DepthPacketStreamParser] 31 packets were lost
> [Info] [TurboJpegRgbPacketProcessor] avg. time: 19.3184ms -> ~51.7642Hz
> [Info] [DepthPacketStreamParser] 31 packets were lost
> [Info] [DepthPacketStreamParser] 30 packets were lost

each line appear every second approximately

r9y9, 2017-05-26:
> [Info] [TurboJpegRgbPacketProcessor] avg. time: 19.3184ms -> ~51.7642Hz

This should be fine?

I think skipping depth packet usually happens and is not a bug. See https://github.com/OpenKinect/libfreenect2/issues?q=is%3Aissue+skip+is%3Aclosed.

Negvet, 2017-05-26:
50 Hz is fine for sure but for some reason the main while loop inside `selective_streams.py` runs just with 3-4 hz (I measured time in python).  For my application, I need at least 10-15 fps.
Plus Protonect works very smooth and gives up to 30 hz, as it seems to me.
Some people experiencing the same kind of issue [#29](https://github.com/r9y9/pylibfreenect2/issues/29)
Do you have an idea?

r9y9, 2017-05-26:
> 50 Hz is fine for sure but for some reason the main while loop inside selective_streams.py runs just with 3-4 hz 

Umm, this is odd. I believe I was getting 15~30 fps with OpenCL packet pipeline on macOS. 

I'd suggest:

1. Try changing packet pipeline (https://github.com/r9y9/pylibfreenect2/blob/288e187ec94fd0df30d483f6c11318820e1d1e06/examples/multiframe_listener.py#L13)
2. Profiling which line of code causes slowdown

Negvet, 2017-05-29:
> Try changing packet pipeline (https://github.com/r9y9/pylibfreenect2/blob/288e187ec94fd0df30d483f6c11318820e1d1e06/examples/multiframe_listener.py#L13)

You were right. The problem was with the pipeline.
Although I am still getting `[Debug] [DepthPacketStreamParser] skipping depth packet` but now the visualization is quite smooth and it looks like getting up to 30 fps.
The exact problem was that i did not install OpenCL as it is optional in [libfreenect2](https://github.com/OpenKinect/libfreenect2) installation manual. 
So I installed OpenCL, then just completely reinstalled `libfreenect2` and `pylibfreenect2` and it works just fine.
Thanks!

closed 2017-05-29 by Negvet