r9y9/pylibfreenect2 #39: Issue with asarray function

yvtheja, 2017-05-31:
Hello all,

When I use the asarray function on the registered frame with the type np.float32, the array is getting resized as 424 X 512. Is it fine? Because we have registered frame as 512 X 424 X 4 and the output of asarray function should restrain dimensions.

If I use the same function with the type np.uint8, the output dimensions are restrained. But since the fourth channel of registered frame would be having the depth frame, in which values range from 0 to 4500, I would like to have the type as np.float32. Any idea how we do that with current version?

Please verify.

Thank you.

r9y9, 2017-05-31:
For now, if you define a frame without `frame_type` ([ref](http://r9y9.github.io/pylibfreenect2/latest/api.html#frame)),  you are responsible to specify correct type as an argument of `asarray`.

```
registered = Frame(512, 424, 4)
arr = registered.asarray(np.uint8)
```

or you can omit `np.uint8` by specifing `frame_type`:

```
registered = Frame(512, 424, 4, frame_type=FrameType.Color)
arr = registered.asarray()
```

These are doing the same thing, but later might look nice. 

> If I use the same function with the type np.uint8, the output dimensions are restrained and the shape function on ndarray shows the dimension as 424 X 512 X 4(Shouldn't it be 512 X 424 X 4 ?). 

This is based on opencv convension, `(height, width, chanenls)`, so should be ok with 424x512x4.

> But since the fourth channel of registered frame would be having the depth frame, in which values range from 0 to 4500, I would like to have the type as np.float32. Any idea how we do that with current version?

How about:

```
a = registered.asarray(np.uint8).astype(np.float32)
```

yvtheja, 2017-05-31:
@r9y9 Thanks for the reply.
Am I correct in assuming the last channel(4th channel) in the registered frame is for depth?
If the registered frame is for color information, why doesn't it has 3 channels?
And if the undistorted frame is for depth, why doesn't it has 1 channel?

r9y9, 2017-05-31:
> Am I correct in assuming the last channel(4th channel) in the registered frame is for depth?

No. I'm not sure why I choose 4 channels, but I now think it should be 3 channels.. (probably I thought it had alpha-channel)

> And if the undistorted frame is for depth, why doesn't it has 1 channel?

It's 1 channel. Use `undistorted.asarray(np.float32)`.

yvtheja, 2017-05-31:
Here it is initialized with 4 channels.
[https://github.com/r9y9/pylibfreenect2/blob/master/examples/multiframe_listener.py#L49](url)

r9y9, 2017-05-31:
That means `bytes_per_pixels`, not channels.  http://r9y9.github.io/pylibfreenect2/latest/api.html#pylibfreenect2.libfreenect2.Frame

r9y9, 2017-05-31:
See also https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Frame.html#af21e69741161ac4f308c04bd3d471dd2

yvtheja, 2017-05-31:
Sorry. Stupid me.
Thanks a lot.

closed 2017-05-31 by yvtheja