r9y9/pylibfreenect2 #41: why can not I find libfreenect2.py in the the pylibfreenect2

wuguangbin1230, 2017-08-22:
Hi,
When I tested the pylibfreenect2 by the anaconda, I got some following errors. I have installed the libfreenect2 correctly, but why could not me find the libfreenect2.py in the file (build/bdist.linux-x86_64)?

runfile('/home/wgb/wgb_install/pylibfreenect2-master/tests/test_libfreenect2.py', wdir='/home/wgb/wgb_install/pylibfreenect2-master/tests')
Traceback (most recent call last):

  File "<ipython-input-1-d4b6ad673d7d>", line 1, in <module>
    runfile('/home/wgb/wgb_install/pylibfreenect2-master/tests/test_libfreenect2.py', wdir='/home/wgb/wgb_install/pylibfreenect2-master/tests')

  File "/home/wgb/anaconda2/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 880, in runfile
    execfile(filename, namespace)

  File "/home/wgb/anaconda2/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 94, in execfile
    builtins.execfile(filename, *where)

  File "/home/wgb/wgb_install/pylibfreenect2-master/tests/test_libfreenect2.py", line 9, in <module>
    from pylibfreenect2 import Freenect2, SyncMultiFrameListener

  File "build/bdist.linux-x86_64/egg/pylibfreenect2/__init__.py", line 108, in <module>

  File "build/bdist.linux-x86_64/egg/pylibfreenect2/libfreenect2.py", line 7, in <module>

  File "build/bdist.linux-x86_64/egg/pylibfreenect2/libfreenect2.py", line 6, in __bootstrap__

ImportError: libfreenect2.so.0.2: cannot open shared object file: No such file or directory

r9y9, 2017-08-22:
It seems like the shared library cannot be loaded. If you installed libfreenect2 in `$HOME/freenect2`,  try:

```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$HOME/freenect2/lib
```

r9y9, 2017-08-22:
What exactly are you trying to do? Where is your `libfreenect2.so.0.2`?

wuguangbin1230, 2017-08-23:
Thank you for your reply!
I would like to run the Dex-Net 2.0  model.
I installed the libfreenect2 in the /home/wgb/libfreenect2 file and my libfreenect2.so.0.2 is in the /home/wgb/libfreenect2/build/lib. And I tried the command, it works. Thank you very much.
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$HOME/libfreenect2/build/lib

r9y9, 2017-08-23:
Glad to hear it worked.

closed 2017-08-23 by r9y9