r9y9/pylibfreenect2 #47: Installing pylibfreenect2 tut

dhruvdesmond, 2018-01-03:
I installed freenect2 from this link https://github.com/OpenKinect/libfreenect2.
When I run your setup it fails . How to change config.h file location beacuse it give config.h not exist.
Protonect example works I checked.
I am sorry for ssuch stupid question but I am a begineer.

r9y9, 2018-01-03:
Seems dup of https://github.com/r9y9/pylibfreenect2/issues/28. See the discussion and let me know if you find a solution.

dhruvdesmond, 2018-01-04:
Still nothing.
I tried python setup.py install.
I get this

> dhruv@dh-7537:~$ python setup.py install
Traceback (most recent call last):
  File "setup.py", line 1, in <module>
    import _pyfreenect2
ImportError: No module named _pyfreenect2

r9y9, 2018-01-04:
Ok, seems dup of #9. Please see my comment https://github.com/r9y9/pylibfreenect2/issues/9#issuecomment-237517523. You are trying to install different package.

dhruvdesmond, 2018-01-04:
How to do this ?
You have set LIBFREENECT2_INSTALL_PREFIX environmental variable (default: /usr/local/).
What is the command.

r9y9, 2018-01-04:
To be sure, are you trying to install **pylibfreenect2**, not pyfreenect2?

dhruvdesmond, 2018-01-04:
yes pylibfreenect2

dhruvdesmond, 2018-01-04:
For skeleton tracking using Kinect2. Any other library are also there for that. If Any please do suggest

r9y9, 2018-01-04:
Google it with "how to set environment variable" to learn how. In short, assuming you are on linux, then type the following command in your terminal:

```
export LIBFREENECT2_INSTALL_PREFIX=${your libfreenect install path}
```

r9y9, 2018-01-04:
libfreenect2 doesn't support skeleton tracking. I'd suggest you to use the official Kinect-SDK or https://github.com/Kinect/PyKinect2.

dhruvdesmond, 2018-01-04:
Thanks. Will look into it.

closed 2018-01-04 by dhruvdesmond