r9y9/pylibfreenect2 #48: Depth-RGB alignment

linwanml, 2018-02-16:
Is there a method to read the depth value from the depth video by using python and opencv?

r9y9, 2018-02-16:
Assuming you are talking about reading depth values from a pre-recorded depth sequence, I think you can just use opencv’s VideoCapture.

linwanml, 2018-02-16:
Yes, I have already read the value with opencv. The depth intensity in opencv is 0-255. Is there a transform equation to map each depth intensity(0-255) into millimeter values?

r9y9, 2018-02-16:
That depends on how you saved depth data. For example, assuming the maximum depth in milli meter was 4500, If you scaled the depth values by 255/4500 and saved it as 8bit integers, then you can recover depth by multiplying 4500/255.

linwanml, 2018-02-16:
Ok, thank you for the answers. I have one more question, and then I will close the issue.
Are there some documents, links or papers that I can have a look about these information?