r9y9/pylibfreenect2 #49: Installation problems

totovr, 2018-03-14:
When I tried to install the `setup.py` file I got the next output:

```
UserWarning: Normalizing '0.1.3-dev' to '0.1.3.dev0'
  normalized_version,
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied
```
***SO Ubuntu 16.04***, ***Python version*** `Python 3.6.3 :: Anaconda custom (64-bit)`

r9y9, 2018-03-19:
Please read the installation guide: http://r9y9.github.io/pylibfreenect2/latest/installation.html

maderafunk, 2018-05-28:
There is something wrong in the newest update regarding OpenCLKdePacketPipeline, I cannot install the newest master.

```
pylibfreenect2/libfreenect2.cpp:1341:17: error: ‘OpenCLKdePacketPipeline’ in namespace ‘libfreenect2’ does not name a type
   libfreenect2::OpenCLKdePacketPipeline *pipeline;
                 ^
pylibfreenect2/libfreenect2.cpp: In function ‘int __pyx_pf_14pylibfreenect2_12libfreenect2_23OpenCLKdePacketPipeline___cinit__(__pyx_obj_14pylibfreenect2_12libfreenect2_OpenCLKdePacketPipeline*, int)’:
pylibfreenect2/libfreenect2.cpp:11740:17: error: ‘struct __pyx_obj_14pylibfreenect2_12libfreenect2_OpenCLKdePacketPipeline’ has no member named ‘pipeline’
   __pyx_v_self->pipeline = new libfreenect2::OpenCLKdePacketPipeline(__pyx_v_de
                 ^
pylibfreenect2/libfreenect2.cpp:11740:32: error: expected type-specifier
   __pyx_v_self->pipeline = new libfreenect2::OpenCLKdePacketPipeline(__pyx_v_de
                                ^
pylibfreenect2/libfreenect2.cpp:11749:96: error: ‘struct __pyx_obj_14pylibfreenect2_12libfreenect2_OpenCLKdePacketPipeline’ has no member named ‘pipeline’
 .pipeline_ptr_alias = ((libfreenect2::PacketPipeline *)__pyx_v_self->pipeline);
                                                                      ^
pylibfreenect2/libfreenect2.cpp: In function ‘void __pyx_pf_14pylibfreenect2_12libfreenect2_23OpenCLKdePacketPipeline_2__dealloc__(__pyx_obj_14pylibfreenect2_12libfreenect2_OpenCLKdePacketPipeline*)’:
pylibfreenect2/libfreenect2.cpp:11814:33: error: ‘struct __pyx_obj_14pylibfreenect2_12libfreenect2_OpenCLKdePacketPipeline’ has no member named ‘pipeline’
     __pyx_t_1 = ((__pyx_v_self->pipeline != NULL) != 0);
                                 ^
pylibfreenect2/libfreenect2.cpp:11822:28: error: ‘struct __pyx_obj_14pylibfreenect2_12libfreenect2_OpenCLKdePacketPipeline’ has no member named ‘pipeline’
       delete __pyx_v_self->pipeline;
                            ^
error: command 'gcc' failed with exit status 1
```

r9y9, 2018-05-28:
Oh sorry, I reverted the change https://github.com/r9y9/pylibfreenect2/commit/78b8748ea356d8ed61d862e24e8eab934d94d6e3. Should be working now..

r9y9, 2018-05-29:
I'm closing this issue. Feel free to reopen or create a new issue if you have further problems.

closed 2018-05-29 by r9y9