r9y9/pylibfreenect2 #50: Apply undistortDepth on numpy array instead of frame object

maderafunk, 2018-05-15:
Hi, 

is it possible to create a function similar to undistortDepth that accepts an numpy array instead of a frame object? My problem is, that I have already saved many depth images as numpy arrays and now I need the undistorted depth arrays.

maderafunk, 2018-05-18:
I solved it by converting the numpy array back as a libfreenect2.frame type. This was possible, as most of the frame data (exposure, gain, etc.) is not needed for the registration functions. 

Would there be any interest integrating that code?

r9y9, 2018-05-20:
Cool. I'd be happy if you can create a PR.

r9y9, 2018-05-22:
I think you forgot to set some of the camera parameters. https://github.com/r9y9/pylibfreenect2/blob/88de3ca61baad4ccaeb6e79d29f160c0ed076ce8/pylibfreenect2/libfreenect2/libfreenect2.pxd#L81, say k1, k2, k3, p1, p2. Those are used in https://github.com/OpenKinect/libfreenect2/blob/83f88b4c09f0b00724ae65785abcd4f3eeb79f93/src/registration.cpp#L73-L85.

As for `# NOTE: must be called after device.start() <----- What does this mean?`, 

https://github.com/r9y9/pylibfreenect2/blob/88de3ca61baad4ccaeb6e79d29f160c0ed076ce8/examples/multiframe_listener.py#L45-L47

This is because camera parameters are retrived after device staretd.

maderafunk, 2018-05-22:
Thanks. I was wondering about those paramters, however device.getColorCameraParams() only returns fx, fy, cx, cy. 

I will try to add the other parameters to cdef class ColorCameraParams and IrCameraParams.

maderafunk, 2018-05-22:
Bam, it works!

r9y9, 2018-05-22:
Nice!

maderafunk, 2018-05-22:
There is only one thing left that's not so nice. When I convert my numpy array back to bytes format so that it can be given to the frame object, it works well when I do that in my python environment with:
`
byte_data = numpy_array.tobytes('C')`

However, I would rather like to handle that internally within pylibfreenect2. But when I use the same function within pylibfreenect2, the conversion is not correct, I can see artifacts in the image. I guess this has to do with cython? Do you know how I could handle that?

r9y9, 2018-05-22:
I think cython can handle numpy array input directly. From one of my projects that uses cython, https://github.com/r9y9/pysptk/blob/abdad498a549c746012418be3a8f264bbaeb9a0e/pysptk/_sptk.pyx#L34-L40 works.

```cython
def acep(x, np.ndarray[np.float64_t, ndim=1, mode="c"] c not None,
         lambda_coef=0.98, step=0.1, tau=0.9, pd=4, eps=1.0e-6):
    assert_pade(pd)
    cdef int order = len(c) - 1
    cdef double prederr
    prederr = _acep(x, &c[0], order, lambda_coef, step, tau, pd, eps)
    return prederr
```

C function signature:

```c
double _acep "acep"(double x, double *c, const int m, const double lambda_coef,
                    const double step, const double tau, const int pd,
                    const double eps);
```

So maybe worth trying something like below?

```
np.ndarray[float maybe?, ndim=2, mode="c"] byte_data
```

and passing the address of it:

```cython
self.ptr = new libfreenect2.Frame(
　width, height, bytes_per_pixel, &byte_data[0]) # we may need reinterpret_cast here
```

r9y9, 2018-05-22:
I will look into it.

maderafunk, 2018-05-22:
Thanks a lot, I will test it tomorrow.

r9y9, 2018-05-23:
That's what I meant by the comment `# we may accept ndim=2 and convert it to ndim=1 here for convenience`. 

I'm guessing relaxing input type from `np.ndarray[np.float32_t, ndim=1, mode="c"] ` to `np.ndarray[np.float32_t, mode="c"] ` and then calling `byte_data.ravel()` or `byte_data.reshape(-1)` just works.

maderafunk, 2018-05-23:
That's it, I didn't understand the comment properly. I added it in the pull request #51

closed 2018-05-23 by maderafunk