r9y9/pylibfreenect2 #52: Record audio with Kinect

sharpriy, 2018-06-25:
Hi guys,

Do pylibfreenect2 provides any interface to record audio stream from Kinect. I have gone through some of the forums and there are some where Kinect ADK is being used for this. One of them is https://social.msdn.microsoft.com/Forums/sqlserver/en-US/f4e79a9f-76c8-4013-a33c-c2b0958806c9/recording-a-continuous-kinect-audiostream-using-cnet?forum=kinectsdk

r9y9, 2018-06-25:
pylibfreenect2 is just a wrapper around https://github.com/OpenKinect/libfreenect2, so you can ask such a feature request to it: https://github.com/OpenKinect/libfreenect2/issues. 

As far as I know there's no interface for recording audio stream at the moment.

sharpriy, 2018-06-25:
Thanks @r9y9