r9y9/pylibfreenect2 #55: pip does not let install pylibfreenect2

AAcquier, 2018-09-13:
Hi, 

My issue seem to be the same one than #36, every time I try to use pip install pylibfreenect2 I get the following message:
`Collecting pylibfreenect2
  Downloading https://files.pythonhosted.org/packages/25/0a/12665fc2a467672ad3daf1823dddbee96ce2b6dfebbb7f3dab97b4a76a84/pylibfreenect2-0.1.2.tar.gz (98kB)
    100% |████████████████████████████████| 102kB 1.3MB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-D9Uer7/pylibfreenect2/setup.py", line 25, in <module>
        raise OSError("{}: is not found".format(libfreenect2_configh_path))
    OSError: /usr/local/include/libfreenect2/config.h: is not found
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-D9Uer7/pylibfreenect2/
`
I am new to python and do not really how to fix as #36 is not really clear on how to fix the problem

AAcquier, 2018-09-21:
I fixed it, I simply found and moved the missing files in the right place

closed 2018-09-21 by AAcquier

zabeehkhan1, 2019-12-13:
can you please show the steps how to move the files in right place . thanks in advance

OancaAndrei, 2023-01-29:
Hi @zabeehkhan1, I've faced the same issue when running `pip install pylibfreenect2`: what solved the issue was exporting the `LIBFREENECT2_INSTALL_PREFIX` environment variable to point to my `libfreenect2` install, as documented [here](https://github.com/r9y9/pylibfreenect2/blob/master/docs/installation.rst#building-pylibfreenect2).