r9y9/pylibfreenect2 #60: Cannot succeed to install on windows10

yzf123456, 2019-04-18:
[image]
I don't understand this? i have installed libfreenect2 correctly and confirm that Protonect works.

r9y9, 2019-04-18:
What’s unclear with you?

yzf123456, 2019-04-18:
i don't know how to set the environmental,because i cannot find /usr/local/ on my computer,
and i want to know if LIBFREENECT2_INSTALL_PREFIX=C:\Users\DELL\Desktop\pylibfreenect2-master is correct? Thank you .

r9y9, 2019-04-18:
That's incorrect. Please do make sure that `${LIBFREENECT2_INSTALL_PREFIX}include/libfreenect2/config.h` exists. Replace `/` to `\` if you use windows.

yzf123456, 2019-04-18:
OK, i find the config.h in C:\Users\DELL\Desktop\libfreenect2-master\build\install\include\libfreenect2,so
i should set the environmental variable LIBFREENECT2_INSTALL_PREFIX=C:\Users\DELL\Desktop\libfreenect2-master\build\install?thank you,i am freshman.

NIravMeghani, 2020-04-11:
[image]
[image]
any solution...?

Selar007, 2021-06-15:
A small guide how to install on windows 10:
https://stackoverflow.com/questions/67902941/how-to-call-multiple-kinect2-device-with-pykinect2/67914523#67914523