r9y9/pylibfreenect2 #64: Depth limited to 4.5 meters pylibfreenect api?

Belidan, 2019-06-07:
Hello,

as the caption mentioned in figured out that in the depth images, the maximum range is 4.5 meters. I need to extend this range to around 8 meters which should be possible. In a post for libfreenct2 it is possible to change the depth by setting the range in the DepthPacketProcessor.  ([Depth limited to 4.5 meters?])(https://github.com/OpenKinect/libfreenect2/issues/143) Is it possible to get access to the DepthPacketProcessor with pylibfreenct2? Maybe i'm just bad at applying the api.

Kind Regards, Dimitrij

r9y9, 2019-06-10:
It's possible but for now you cannot change depth limit since the python API doesn't expose getDepthPacketProcessor method. https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1PacketPipeline.html

You can try exposing the API from the python binding if you want the feature.

Belidan, 2019-06-18:
Ah thank you very much. Do you have a nice tutorial how to expose those APIs? I'm quite new to Python and i guess this is a good to know.

r9y9, 2019-06-18:
You could look at the cython doc: https://cython.readthedocs.io/en/latest/

Archeema16, 2020-01-12:
Hi. I am working on Kienct V2 with pylibfreenect2 . I need to measure an object distance from Kinect in mm/cm etc. I am trying to access Raw depth data for the last one month but i am not able to access it or even get sense out of things i get. Its like entering from one rabbit hole to another. Can you help me on it.

r9y9, 2020-01-12:
I am not sure what you mean by you cannot access raw depth data. Raw depth data is surely available. See https://github.com/r9y9/pylibfreenect2/blob/12c405e187f19fa77887d4ba16eb1241e75bb2d5/examples/multiframe_listener.py#L77 for example. Measuring distance for objects requires object detection algorithms though (which is not supported in either libfreenect2 or pylibfreenect2).

Archeema16, 2020-01-12:
You are right object detection is another thing and I am not talking about.it. 
 cv2.imshow("depth", depth.asarray() / 4500.)  This line shows you a depth frame. 
But what i need is to get real life distance of a pixel Kinect v2 is showing me. And i want that distance in mm/cm. So how to convert depth frame or use depth frame to get that particular pixel(X,Y) depth in mm/cm.

A point i will like to add here while searching i came across function GetPointXYZ and i thought it gave real life Z(distance) from camera to a particular pixel. But it only showed me NAN.Here is the code of it.  x and y are my required pixel coordinates.
	out = np.zeros((1,3), dtype=np.float64)
	if x!=0 and y!=0:
		X, Y, Z = registration.getPointXYZ(undistorted, x, y)
		out[0] = np.array([Z, X, -Y])
		print(out)

r9y9, 2020-01-12:
getPointXYZ(implemented in https://github.com/r9y9/pylibfreenect2/pull/23) should work. Please note that you need to initialize and start your kinect2 device before calling getPointXYZ, since registration requires to retrieve camera parameters from kinect2.

Another option is to implement the conversion by yourself. It's just simple math.