r9y9/pylibfreenect2 #70: Creating Frame from numpy array

buxbomber, 2020-03-09:
Hi ,

I need to create point clouds from existing depth/color data. To do this, I figured I needed to create some Frames with the data. I am trying to initialize a Frame instance with a 2D numpy array of depth data, but I keep getting errors (like "ndarray is not C-contiguous"). Could someone help me out with the syntax on this? Or a better way of converting existing numpy arrays of depth data into Frames/point clouds? Thanks!

buxbomber, 2020-03-09:
Solved it by using: Frame(512, 424, 4, -1, np.ascontiguousarray(np_array,dtype=np.float32)) where np_array is your 2d depth numpy array

closed 2020-03-09 by buxbomber

MengHao666, 2020-06-18:
> Solved it by using: Frame(512, 424, 4, -1, np.ascontiguousarray(np_array,dtype=np.float32)) where np_array is your 2d depth numpy array

Hi,buxbomber.

I also want ro create point clouds using this repo.
And I use **Undistorted depth frame** and  **Registered color frame** two data. However， how should I get the camera internal parameters？I use this two data cause I think all of them have been processed by registration.

Do u have any idea about that?
I could be very grateful if u could share.