r9y9/pylibfreenect2 #75: is Raspberry pi 4 support?

Jyurineko, 2020-05-13:
thx for amazing work,
but may i ask is Raspberry pi 4 support?
i follow the document and after "pip install pylibfreenect2"
it shows
______________________________________________________________________________
root@raspberrypi:/home/pi/Desktop/CGPraktikum/pylibfreenect2# LIBFREENECT2_INSTALL_PREFIX=/home/pi/Desktop/CGPraktikum/libfreenect2 pip install pylibfreenect2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pylibfreenect2
  Using cached https://files.pythonhosted.org/packages/0d/08/c006596cd6b5802cdb9da3f703ad0d8828dedc57c7bfece46d1524d543de/pylibfreenect2-0.1.3.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-oZBhki/pylibfreenect2/setup.py", line 35, in <module>
        raise OSError("libfreenect2 library cannot be found")
    OSError: libfreenect2 library cannot be found
______________________________________________________________________
i tried method in another issue like : `export LD_LIBRARY_PATH=$HOME/freenect2/lib:$LD_LIBRARY_PATH`

or in setup.py add `libfreenect2_install_prefix = os.environ.get(
    "LIBFREENECT2_INSTALL_PREFIX", "/home/pi/Desktop/CGPraktikum/libfreenect2")`

but it still same problem. 
so can you give some advice? 
thx in advance:)

r9y9, 2020-05-14:
I think it should work. Please make sure you have a libfreenect2.so or similar in your file system. 

https://github.com/r9y9/pylibfreenect2/blob/981d62a93dab3c9bd8b231d1239dad798cddc8f1/setup.py#L27-L35

Jyurineko, 2020-05-14:
thx for reply, the setup.py install is successful. 
i forget my is LD_LIBRARY_PATH=/root/freenect2 

may i ask should i write OpenGL ES pipeline by myself?
or what should i do can create a visual  window?
because everytime i run it shows 
------------------------------------------------------------------
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 GLX: Failed to create context: GLXBadFBConfig
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
-------------------------------------------------------------------
sorry i am newbie :)

r9y9, 2020-05-17:
You should first check if libfreenect2 works okay on your platform.

MikoyChinese, 2020-05-20:
@Jyurineko OpenGL pipeline need a monitor to run, or you can build a headless opengl library. I guessed that you run this on ssh tunnel without X forward, so you can try run this on other pipeline such as opencl or cpu or cuda. These all can work on host without monitor.

Jyurineko, 2020-05-20:
@MikoyChinese i used VNC to remote the monitor, i try direct connect a monitor to run today,
but i also tried ssh -X pi@ip address. and using VNC still same error:(
____________________________________________________________________________
pi@raspberrypi:~/Desktop/CGPraktikum/freenect2-python/examples $ sudo DISPLAY=0 python3 ir_camera.py
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 014443550647
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 X11: Failed to open display 0.0
[Error] [OpenGLDepthPacketProcessor] Failed to initialize GLFW.

MikoyChinese, 2020-05-20:
@Jyurineko :), Hah, If you want to use opengl pipeline you need a monitor or headless opengl library. But I just think that you can just use CPU pipeline or OpenCL pipeline enough.

example code:

```python
from pylibfreenect2 import Freenect2, SyncMultiFrameListener
from pylibfreenect2 import FrameType, Registration, Frame
from pylibfreenect2 import OpenCLPacketPipeline

pipeline = OpenCLPacketPipeline()

enable_rgb = True
enable_depth = True

fn = Freenect2()
num_devices = fn.enumerateDevices()
if num_devices == 0:
    print("No device connected!")
    sys.exit(1)

serial = fn.getDeviceSerialNumber(0)
device = fn.openDevice(serial, pipeline=pipeline)

types = 0
if enable_rgb:
    types |= FrameType.Color
if enable_depth:
    types |= (FrameType.Ir | FrameType.Depth)
listener = SyncMultiFrameListener(types)

# Register listeners
device.setColorFrameListener(listener)
device.setIrAndDepthFrameListener(listener)

if enable_rgb and enable_depth:
    device.start()
else:
    device.startStreams(rgb=enable_rgb, depth=enable_depth)
```