r9y9/pylibfreenect2 #81: Issues running examples

Dofinncng, 2020-11-07:
Hello, 
the installation of the module pyllibfreenect seemed to have worked fine, but when trying to run the examples provided an error occurs. 
The import itself seems to fail, like you can see in the picture:
[image]

The error message that occurs when trying to run the progam is the following one:
[image]

Has anyone ever run into that problem and could help? Would be much appreciated.

Dofinncng, 2020-11-07:
"Das angegebene Modul wurde nicht gefunden" means: The specified module could not be found

SizheAn, 2022-02-17:
same problem exactly... Use dumpin/dependencieswalker to check and found that I had all the .dll files in my PATH... But still have this problem. Any help?